smartbutnot/flightportal #4: Setup help

Tigwin, 2023-01-20:
Hello, I bought the components and am excited to see if I can get this working.

This is what I did so far:

- Assemble the hardware.
- Plug the M4 to my PC. 
- Double click to enable debug
- Upload adafruit-circuitpython-matrixportal_m4-en_US-7.3.3.uf2
- CIRCUITPY drive appears on my desktop
- uploaded code.py to root folder
- uploaded secrets.py to root folder, after changing wifi name and pass
- Entered 28.021682,27.887032,-82.553315,-82.310081 to bounds box (trying to draw around Tampa International Airport for testing)

Unplug and plug back in the M4. 

It has a green light on the rear, but the display is black. I don't know what it's doing. Any way to debug what's happening, or did I miss a step somewhere?

Thanks!

smartbutnot, 2023-01-20:
Sounds like you might have missed adding the libraries? https://learn.adafruit.com/adafruit-matrixportal-m4/circuitpython-setup

In any case, you can use a serial connection to see what errors are being produced, see the bottom of the read me file.

Tigwin, 2023-01-20:
Ah, you're right, I downloaded the bundle but didn't upload it because I misread this comment:

> The libaries it needs are I think all part of the recommended prep above, but for info they are:

I thought that meant the uf2 file contained it.

So I grabbed adafruit-circuitpython-bundle-8.x-mpy-20230120.zip and copied the lib/ folder to the lib/ folder on the CIRCUITPY drive.

Once files were uploaded, I restarted the device. The light now shines blue (didn't before).

Now it works!

Thank you.... I had this sitting on my desk for a few weeks and didn't proceed because I thought the uf2 files was some kind of mac software (based on the screenshots) and the download links all seemed to not have executables I could run. I downloaded Arduino and some other stuff I didn't have to.

Tigwin, 2023-01-20:
Any idea about this error:

Reply is OK!
New flight 2eeb8478 found, clear display
Details lookup saved 5866 bytes.
JSON error
'NoneType' object isn't subscriptable
error parsing JSON, skip displaying this flight

Tigwin, 2023-01-20:
All fixed

closed 2023-01-20 by Tigwin