smartbutnot/flightportal #5: Failures when fetching

Tigwin, 2023-01-20:
My display pulls and displays flights successfully, but there are some that fail. 

Any idea about this error, I've seen it happen 3 times so far.

```
Reply is OK!
New flight 2eeb8478 found, clear display
Details lookup saved 5866 bytes.
JSON error
'NoneType' object isn't subscriptable
error parsing JSON, skip displaying this flight
```

I pulled up the flighttracker url, and next time it happens I'll update this post with what it is returning when this happens.

smartbutnot, 2023-01-20:
The error handling is fairly basic and if something expected is missing from the json, it skips it. If it’s happening often you could add in a statement to print the whole json object and see what’s wrong with it…

Tigwin, 2023-01-21:
`New flight 2eed0f43 found, clear display
Details lookup saved 1989 bytes.
JSON error
'NoneType' object isn't subscriptable
error parsing JSON, skip displaying this flight`

`{"full_count":14347,"version":4,"2eed0f43":["A0B45C",25.2000,-70.4167,89,3700,85,"","T-F5M","C150","N1443Q",1674314719,"ZPH","","",0,0,"N1443Q",0,""]
}`

smartbutnot, 2023-01-25:
When you do a details lookup on 2eed0f43 it looks like it's missing things like a flight number, which is why the function errors out. Is it happening for many flights? So far I've just had it skip flights with missing data, if people see flights like that some significant % of the time then I'll improve the handling.

Tigwin, 2023-01-25:
I'm not sure if they are airforce or what, but I've seen a number of airforce planes on the flight24 website, and private planes. There's some that say "Private" in the json, so not sure what this specific plane is doing. I have seen at least a half dozen when I was watching. No idea how common it is.

Tigwin, 2023-01-30:
Hello, here's the latest error

`Get flights...
Retrieving data...Reply is OK!
No flights found, clear display
memory free: 49184
Get flights...
Retrieving data...Reply is OK!
New flight 2f056964 found, clear display
JSON error
'NoneType' object isn't subscriptable
error parsing JSON, skip displaying this flight
memory free: 49168
Get flights...
Retrieving data...Traceback (most recent call last):
  File "[code.py](http://code.py/)", line 385, in <module>
  File "[code.py](http://code.py/)", line 361, in get_flights
  File "adafruit_portalbase/__init__.py", line 450, in fetch
  File "adafruit_portalbase/network.py", line 586, in fetch_data
  File "adafruit_portalbase/network.py", line 505, in fetch
  File "adafruit_requests.py", line 820, in get
  File "adafruit_requests.py", line 658, in request
  File "adafruit_requests.py", line 340, in close
ValueError: invalid syntax for integer with base 16

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Matrix Portal M4 with samd5                                                                                  1j19
>>>`

Attached is the code that was running too.

[code.py.txt](https://github.com/smartbutnot/flightportal/files/10538040/code.py.txt)

dranovsky, 2023-01-30:
My flightportal seems to die regularly, the last flight to kill it was 2f062a05 which seems military.
After this, it fell off wifi and wouldn't reconnect at all until a manual reset.

smartbutnot, 2023-01-30:
This one: 

`ValueError: invalid syntax for integer with base 16`

Seems like it's requests somehow getting into a strange socket state and the connection throwing errors. The good news is that just adding "ValueError" to the exceptions caught should handle it gracefully, will do that now. Remains to be seen if everthing works OK on the next loop, of course.

The connection loop I'm not sure what the situation is, the matrixportal is detecting that it's not got a network connection and trying to reconnect - that's all core adafruit code, so it's not something I can mess with. There's some code here that prints out signal strength, could be worth checking your portal has a solid connection: https://learn.adafruit.com/adafruit-matrixportal-m4/internet-connect

closed 2023-02-07 by smartbutnot

Tigwin, 2023-03-06:
JFYI - been running this and there are no more errors, it works great. Thank you