smartbutnot/flightportal #6: Flip display 180 degrees?

Tigwin, 2023-01-20:
I checked the code but didn't see a way to rotate the display. I'm planning to mount the display on the wall and the power cable would be better off to the left instead of the right. Any way to flip it?

ty

smartbutnot, 2023-01-20:
If you use matrixportal.graphics as described here, https://learn.adafruit.com/creating-projects-with-the-circuitpython-matrixportal-library/choosing-your-modules , then graphics seems to have a rotation argument as you can see here https://github.com/adafruit/Adafruit_CircuitPython_MatrixPortal/blob/main/adafruit_matrixportal/matrixportal.py

smartbutnot, 2023-01-25:
Actually, it turns out the matrixportal itself has a rotation argument! I've added it to the code now, change "rotation=0" to 180 and it should do what you want.

closed 2023-01-29 by smartbutnot