smartbutnot/flightportal #7: Two things, speed+altitude, and a parse error

Tigwin, 2023-01-25:
I was messing around with code.py and I like adding these two values, they draw initially then get overwritten, so no details are lost from the display:

display_flight():
```
    matrixportal.display.show(g)
    label1.text=label1_short
#    label2.text=label2_short
#    label3.text=label3_short
    label2.text=label4_long
    label3.text=label4_short
```

parse_details_json():

```
	label4_short=str(altitude)+" ft"
	label4_long=str(speed)+" mph"

```

Also I'm getting parse errors randomly, is there a way to auto reload or something? This error causes the app to stop running, then it hangs:

```
Get flights...
Retrieving data...Reply is OK!
No flights found, clear display
memory free: 49424
Get flights...
Retrieving data...Reply is OK!
New flight 2ef7a4d3 found, clear display
Traceback (most recent call last):
  File "[code.py](http://code.py/)", line 388, in <module>
  File "[code.py](http://code.py/)", line 236, in get_flight_details
NameError: local variable referenced before assignment

Code done running.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disab

```
[code.py.txt](https://github.com/smartbutnot/flightportal/files/10503122/code.py.txt)

smartbutnot, 2023-01-25:
Hmm - is this your own code? Those line references don't line up with anything, and reference before assignment should mean something weird happening with a variable. But no, no way to get it to reload on a crash, that I know of.

Tigwin, 2023-01-25:
I made a few random small changes as described above - the file is attached above as code.py.txt if you want to look.

smartbutnot, 2023-01-29:
Hmm, it's somehow getting to the end of the big json without finding the end of the strings it wants ("trail_end"). I'll add in a lione to handle that gracefully, lilke this:

if trail_end:
        print("Details lookup saved "+str(trail_end)+" bytes.")
        return True
    else:
        return False

Tigwin, 2023-01-29:
Thanks, I added it. Will report back if it still has the error.

Tigwin, 2023-02-04:
The trail_end code was added but I'm got this error a second time (appears to be identical to the one I posted on the other issue):

JSON error
'NoneType' object isn't subscriptable
error parsing JSON, skip displaying this flight
memory free: 49344
Get flights...
Retrieving data...Traceback (most recent call last):
  File "[code.py](http://code.py/)", line 385, in <module>
  File "[code.py](http://code.py/)", line 361, in get_flights
  File "adafruit_portalbase/__init__.py", line 450, in fetch
  File "adafruit_portalbase/network.py", line 586, in fetch_data
  File "adafruit_portalbase/network.py", line 505, in fetch
  File "adafruit_requests.py", line 820, in get
  File "adafruit_requests.py", line 658, in request
  File "adafruit_requests.py", line 340, in close
ValueError: invalid syntax for integer with base 16

Code done running.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

smartbutnot, 2023-02-04:
Try the most recent version of the code, it should now catch a ValueError if one is thrown in get_flights. Hopefully it'll just move on and work correctly next loop - seems to be a case of requests getting into a strange state with the socket...

Tigwin, 2023-02-04:
Will try it

Tigwin, 2023-02-06:
Woke up this morning and the tracker had stopped. Here's what it showed using your latest (unaltered) code:

No flights found, clear display
Retrieving data...Reply is OK!
No flights found, clear display
Retrieving data...Reply is OK!
No flights found, clear display
Retrieving data...Reply is OK!
New flight 2f161a80 found, clear display
Traceback (most recent call last):
  File "code.py", line 370, in <module>
  File "code.py", line 229, in get_flight_details
NameError: local variable referenced before assignment

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

smartbutnot, 2023-02-06:
Ah, thanks for the stacktrace. I think logically the only way that can hit is if it successfully gets loads all the JSON without running out of space (it would throw a graceful "exceeded max size" error for that), but doesn't find the start of a trail entry, which means it never even tries to find the end of a trail entry, which means it errors out when checking the value of trail_end because it never got set to anything. So I need to add some code which handles "get through all the JSON without finding anything".

Weird thing though - if I force my portal running the same stock code to return that same flight ID and parse it, it gets through the JSON fine. So I guess it must be something weird/wrong with the way the JSON got returned on that particular request, not the data itself.

But anyway - new code going up now, which should handle that error if it occurs again (best I can figure).

Tigwin, 2023-02-07:
Great... trying again now

closed 2023-02-07 by smartbutnot