smartbutnot/flightportal #8: Grabbing Multiple Flights in Area

jacobjdev, 2023-01-29:
Hello,
I believe there may be an issue where if you have multiple aircraft in the box, it seems the code currently picks one, (maybe returning the top result in a list?) and sticks with it until it leaves the bounding box, instead of cycling through the other aircraft.  I'm not sure if this is a current design limitation.  

Current way to replicate it:  Set bounding box to a major airport with multiple aircraft in the area, and in the debug messages it says it keeps finding the same aircraft.

Thanks for looking into this!

smartbutnot, 2023-01-29:
Yeah, the way it works at the moment is to always ask "what's the first flight in this area", with fr24 deciding what constitutes first. I'll think about it, it would complicate things a bit to hold state on recent flights seen but that is doable.

jacobjdev, 2023-01-29:
Thank you for considering!

closed 2023-01-29 by jacobjdev