smartbutnot/flightportal #9: Infinite WiFi unsuccessful WiFi reconnect attempts after connection drops

smartbutnot, 2023-02-07:
My flightportal seems to die regularly, the last flight to kill it was 2f062a05 which seems military.
After this, it fell off wifi and wouldn't reconnect at all until a manual reset.

Retrieving data...Reply is OK!
New flight 2f062a05 found, clear display
Details lookup saved 5574 bytes.
JSON error
'NoneType' object isn't subscriptable
error parsing JSON, skip displaying this flight
Connecting to AP xxxxxx
Could not connect to internet ('Failed to connect to ssid', b'xxxxxx')
Retrying in 3 seconds...
Connecting to AP xxxxxx
Could not connect to internet ('Failed to connect to ssid', b'xxxxxx')

(repeat, ad infinitum)

_Originally posted by @dranovsky in https://github.com/smartbutnot/flightportal/issues/5#issuecomment-1409368771_

smartbutnot, 2023-02-07:
Just had this happen on my reference setup so definitely an issue, and not unique to the WiFi network above. It's in the adafruit portalbase code at this point so hard to diagnose. Should be manageable though - planning to access the network layer and do some WiFi reset checks, which should reboot the whole ESP32 part of the board. Can also add a watchdog check to reboot the whole system, but have been avoiding that so far as it could mask all sorts problems.

medd49, 2023-09-08:
Anyone have a solution for this? Am having the same issue - though my matrixportal seems to eject after a failed connection attempt in an infinite loop, making for an even more frustrating debugging process

smartbutnot, 2023-10-22:
That's probably the watchdog resetting the board (this is what the w.feed stuff in the code is doing, keeping it alive). It's a bit of a catch-22, either try and do as low-level a reset as possible to have the best chance of fixing a wifi issue, or keep the board alive so you can debug, but increase the chance that it gets stuck. If you want to debug, you can comment out the watchdog parts. From what I can tell, sometimes the matrixportal just gets stuck on wifi and all you can do is power cycle it.