smartbutnot/flightportal #10: Off-colored pixels in first few rows

felker, 2023-02-22:
Thanks so much for making this project public!

Just wondering if you have ever seen the following phenomenon before: when a flight appears and the screen is updating, some pixels in the first few rows deviate from the intended white color and flicker green and red. I assume it is a problem with a connection or my LED panel from AliExpress, but thought I would ping you to check if you might have an idea before I replace the panel. Cheers!

[image]

felker, 2023-02-27:
I tried it on a spare panel (both from https://www.aliexpress.us/item/3256804589638597.html) and had similar, but not identical, behavior. The photo looks a bit different since there is no diffuser panel on the spare LED matrix.

[image]

For the original panel:
- Flickering pixels only (largely?) happen in the first few rows, with the Matrix Portal plugged directly into the right side `DATA_IN` socket (when viewed from the front of the panel). Since I don't use an IDC cable I don't think that connection is loose, but maybe a loose power connection in either the fork terminators on the Matrix Portal, or the USB-C input, or the Molex-style power-in connector on the panel? Everything seems to draw < 0.7 A at peak. 
- 95% of the flickering happens when either the first two rows of text (flight number and airline, or Origin-Destination) are moving and reach the "problematic region" on 1/3 of the left side. Slight flickering again when the panel is about to go dark.
-  Curiously, the motion of the bottom row text (airplane model) does not seem to cause any flickering.

Will play around with the rotation and scroll speed variables and report back.

closed 2023-02-27 by felker

felker, 2023-02-27:
Replaced USB-C power input brick and uninstalled/reinstalled Matrix Portal and forked terminal power connectors, and the flickering in the first 3 rows got worse. Also seeing flickering in row 8 now... stumped

felker, 2023-03-03:
This is what it looks like with rotation set to 180; the flashing/ghosting and/or static off-colored pixels have definitely moved from the top left to the center/top-right. The degree to which it happens is somewhat variable. 

https://user-images.githubusercontent.com/1410981/222661278-222c6fec-d257-4365-8bb6-a482f6480948.MOV

[image]

smartbutnot, 2023-03-05:
That second picture looks like you have ghosting of some of the pixels from the row below, shifted 8 pixels up. Can't really say what would cause it, I've had similar things happen with bad connections on other hardware in the past, but there's not much to go wrong between the matrixportal and the socket...