smartbutnot/flightportal #11: NeoPixel Yellow status?

felker, 2023-02-27:
According to https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/410e5a8b11e7ead6dd5c4293070b6b6d7e64a5a3/adafruit_esp32spi/adafruit_esp32spi_wifimanager.py#L45-L50 the NeoPixel turns Red if there is no WiFi connection, Green when a connection is established. 

For me, the NeoPixel is typically solid green after startup, but occasionally flashes yellow. How should I interpret this status?

felker, 2023-03-10:
I also asked on the [Adafruit forums](https://forums.adafruit.com/viewtopic.php?p=964070), and they pointed me to the following lines in the `PortalBase` library:

https://github.com/adafruit/Adafruit_CircuitPython_PortalBase/blob/2085495f4ebc0446287f74af26e36f937a6ea379/adafruit_portalbase/network.py#L59-L66

In particular:
```
STATUS_FETCHING = (150, 100, 0)  # Orange
```
Which I suppose looks somewhat yellow-orange to me; even in the [comments](https://github.com/adafruit/Adafruit_CircuitPython_PortalBase/blob/2085495f4ebc0446287f74af26e36f937a6ea379/adafruit_portalbase/network.py#L516) they call it yellow later on:
```
            self.neo_status(STATUS_FETCHING)  # yellow = fetching data
```

I guess I am still wondering why the other status colors from PortalBase are never triggered, e.g. `STATUS_DOWNLOADING` or `STATUS_CONNECTED` for blue and cyan? 

Your code.py basically uses only `requests.get()` for network functionality, as I understand it. So perhaps https://github.com/adafruit/Adafruit_CircuitPython_Requests/blob/main/adafruit_requests.py is somehow built on the PortalBase, but Adafruit_CircuitPython_ESP32SPI is independent of PortalBase, so the `connect()` function there for WiFi is never used and ` self._wifi.neo_status(STATUS_CONNECTED)` (blue) is not called?

[image]

smartbutnot, 2023-10-22:
I never really paid much attention to the neopixel because I'm using a case that blocks it - but the code is using adafruit's implementation of requests directly, rather than the built-in matrixportal fetch method (I'm sure there was a good reason for this, but I can't remember what it was). So I think it's just the wifi code that has a reference to the status light, which is this: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/main/adafruit_esp32spi/adafruit_esp32spi_wifimanager.py

But that only has code to set the neopixel to R, G or B. So... yeah, no idea :)