smartbutnot/flightportal #14: Idea for the project

Tigwin, 2023-05-08:
My flight portal is working great, but I was thinking about moving out of my office and into my back yard by my pool. That way if I hear a plane and see a plane, I can easily learn more about it. Right now I have a clock mounted on a wall by the pool, and I was thinking it would be great if the matrix display would show the time when no flights are seen, that way it can have dual purpose. 

Dumb or good idea?

Tigwin, 2023-05-19:
I used this to get the current date: https://learn.adafruit.com/adafruit-funhouse/getting-the-date-time

In clear_flight() I'm calling:

    
    label1.text=response.text
    label2.text=label1.text.substring(10,8)
    label3.text=""

The reply string is too long to show, so label1.text will contain the date, and I tried to grab the time with substr for label2.text but it craps out.

smartbutnot, 2023-10-22:
Nice. Yeah, you could put anything you wanted in at that point really and it will run it/show it whenever it doesn't find a flight.

PG20050223, 2024-05-27:
Note that in your example 'X-AIO-Key' is case sensitive.

Thanks for the example, it works well, and is a cool addition to the Flightportal