smartbutnot/flightportal #16: Matrix Portal restarts when Code file added

putz57, 2023-08-25:
I have this set up for my father in law and he's been loving it, but it stopped working a few months ago and it seems that the matrix portal is continuously restarting after a few seconds of being plugged into the computer or power supply.  

I have gone and deleted the contents of the matrix portal and started fresh which stopped the constant restarting, however as soon as I add the Code.py file, it begins the cycle of restarting again.  This stops again when I delete the Code.py file.  The secrets file does not seem to be causing this issue.  

I have version 8.2.4 installed on my board and a did a fresh download of the required libraries.  

Thanks for a great design and any help you may be able to provide.

smartbutnot, 2023-10-22:
If you setup the Arduino IDE like this: https://learn.adafruit.com/adafruit-matrixportal-m4/arduino-ide-setup you can use the serial monitor to see if the board is saying why it's restarting

iminspaaace, 2024-12-02:
I am having the same issue - did you make any discoveries? I'm on Circuit Python 8.2.10