smartbutnot/flightportal #19: Using a Adafruit MatrixPortal S3

CyberGhostorangefish, 2023-12-26:
Hi, I'm trying to get this project working but I get nothing on the LED panel and no errors on the serial, do you think any code changes are needed to make it work with an S3 rather than an M4

smartbutnot, 2023-12-29:
Oh cool, there's a new version! The extra memory should really help, I ran out pretty fast on another project.

I imagine the wifi manager might be different for the different chip, and the "board" setup at the top of the code must be different. All you need is an S3 circuitpython example to crib from and all the functional code should work the same. Can't actually find one myself.

CyberGhostorangefish, 2023-12-29:
Thanks for the reply, I've given it a go but I get:

code.py output:
Traceback (most recent call last):
File "code.py", line 72, in
AttributeError: 'module' object has no attribute 'ESP_CS'

Any ideas

smartbutnot, 2023-12-29:
Looks like solomonreal has done an S3 fork: https://github.com/solomonreal/flightportal

CyberGhostorangefish, 2023-12-29:
Works like a charm thanks!

closed 2023-12-30 by smartbutnot