smartbutnot/flightportal #23: Board resets after finding a flight

medd49, 2024-06-27:
I have been running this for about a year now with no issues. Recently, I've found that the board stays connected whenever there is no flight overhead. However, once the board does find a flight, it will display it correctly, but then crash during the next loop. This ONLY happens after it successfully found a flight. It seems to have an issue with the following line: 

```python
response=requests.get(url=FLIGHT_SEARCH_URL,headers=rheaders).json()
``` 
in the get_flights definition. 

The strange thing is that it works fine the first time it finds a flight, but then resets when trying to do so again. It'll then reconnect to the wifi via the CheckConnection Function and find the next flight, only to reset again after finding that one.

PG20050223, 2024-06-29:
I am having this issue also. Same as described above.

medd49, 2024-06-29:
@PG20050223 Yes! Same here - about 7 days ago. Glad to know its not just localized to my setup. I've tried debugging some, but once I tracked it to the JSON parsing, that is above my paygrade 😄 

Hopefully someone is able to figure out the issue. I'll keep trying

resmithn, 2024-09-19:
Also same issue here, just noticed a few weeks ago after I plugged board into PC. [Normally runs on a USB Wall plug] I tried adding w.feed after each scroll but no difference. Let it run with MU serial open and don't see any error msgs, e.g. HTTP. The board just disconnects so hard to see/know what the error may be. Of course reconnects and works if flight is found. Wanted to get in the loop if a solution is found.

resmithn, 2024-09-22:
[StarTrailMeteor] opened an issue re: running Flight code on an S3. I had a Word Clock running on an S3 so changed the wifi and request code in the current flight code to what the Word Clock S3 was using and loaded the flight code on another S3. It works and the bigger surprise is: No disconnects, none.

PG20050223, 2024-09-22:
Awesome! Can you share the working World Clock wifi and request code?

resmithn, 2024-09-22:
[PG20050223]: I would but this thread is for the M4 Portal. [StarTrailMeteor] was trying to run on ESP32-S3 so that mod code is what I sent to [StarTrailMeteor] . It won't work on an M4 Portal. Still trying to figure out the disconnect issue on the M4, will add comments when/if I [or someone] can figure out the issue.

compileByTheMile, 2024-12-29:
Moving to CircuitPython 9.x fixed this for me https://github.com/smartbutnot/flightportal/pull/34

medd49, 2025-01-03:
Me too! Moving to v9 seemed to do the trick.

closed 2025-01-03 by medd49