smartbutnot/flightportal #26: Need help

MarquisMCMXCIV, 2024-09-18:
I know nothing about programming and ordered all the parts thinking I would figure it out, but it's been a week and I can't find a way to make it work. I followed all the steps from adafruit, transfered the libraries onto the board, copied the code.py and secrets.py (with my information in it) but I always get error messages. I know it is connected to my wifi, but from there I don't know how to diagnose the problems I am facing. 

A desesperate pilot that just wants a cool addition to his living room decor..

Thanks for the help

resmithn, 2024-09-19:
What are the error messages you are seeing?

MarquisMCMXCIV, 2024-09-19:
Traceback :
  File "code.py", line 144, in <module>
AttributeError: .show(x) removed. Use .root_group = x

And after saving the code.py on the board, it keeps resetting the matrix portal non stop.

resmithn, 2024-09-19:
Below are the lines before/after line 144 in code.py. I think "show" has
changed in CircuitPython 9x, so question is what version are you using on
the M4.
Found this on Github which may help, here is the link:
https://github.com/adafruit/circuitpython/issues/8499
And the comment within was this:

show is still valid in 8.x and will be the last version it is valid.
transitioning to total deprecation of display.show in 9.x in favor of
display.root_group
do something like this instead

main_group = displayio.Groupdisplay.root_group = main_group

If you are in fact using Circuitpython 9.x then try changing line 144 to:
display.root_group = g

lines from code.py below

139 label3_long=''
138
139 # Add the labels to the display
140 g = displayio.Group()
141 g.append(label1)
142 g.append(label2)
143 g.append(label3)
144 matrixportal.display.show(g)
145
146 # Scroll the plane bitmap right to left (same direction as scrolling
text)
144 def plane_animation():

resmithn, 2024-09-19:
this is corrected line, got concatenated in email
main_group = displayio.Group
display.root_group = main_group

resmithn, 2024-09-19:
Just an additional comment. I am using Circuitpython 8.1.0 . I tried 9.x with this code and received a lot of errors. You may want to consider loading the uf2 file for 8.1.0 and also use the libraries for that version.

here is a link to download from Adafruit:
https://circuitpython.org/board/matrixportal_m4/ 
Scroll down to Previous versions.

resmithn, 2024-09-19:
> Don't know if I did all the steps you said correctly, but if so, those are
> the messages I get now.
>
> Traceback (most recent call last):
> File "code.py", line 5, in
> File "adafruit_matrixportal/matrixportal.py", line 33, in
> File "adafruit_matrixportal/network.py", line 32, in
> File "adafruit_portalbase/network.py", line 28, in
> File "adafruit_io/adafruit_io.py", line 30, in
> File "adafruit_minimqtt/adafruit_minimqtt.py", line 37, in
> ImportError: no module named 'adafruit_connection_manager'

You didn't mention what version of CircuitPython you loaded. I added some
additional comments about CP 9.x and this code not working..
I think this project was written for CircuitPython 8.x or maybe even 7.x so
I would suggest you load CircuitPython 8.1.0 as I mentioned in the
subsequent comments I made.
That is what I have loaded on the M4 and it works okay, aside from the
other issues that are in the issues thread, i.e. board resets.

MarquisMCMXCIV, 2024-09-19:
Like I said, I was using latest vesrsion which was 9.X and now I tried puting 8.1.0 like suggested, same for the libraries and now I get the messages previously sent. Not even sure if all I did, while "trying" to install 8.1.0 was correctly done as I am a newbie to programming and very limited in my knowledge..

Sorry if what I say doesn't make sense to you

resmithn, 2024-09-19:
This is what I have loaded in the lib folder for 8.1.0
[image]

MarquisMCMXCIV, 2024-09-19:
yes, got the libraries from the link you sent me and just added the missing libraries to my board.

now that's what I got

Code stopped by auto-reload. Reloading soon.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Check and reconnect WiFi
Connect attempt 1 of 10
Reset ESP...
Attempt WiFi connect...
Successfully connected.

Does it mean it's working and now I just need to wait for a plane to fly over?? :)

resmithn, 2024-09-19:
Looks Good!! Look in the code.py around line 419 for 
#print("Get flights...")
 flight_id=get_flights()
 w.feed()
remove the # from the print statement, just a visual that you got that far.
Comment if you need anything further.

MarquisMCMXCIV, 2024-09-19:
Ok I'll do that, thanks!

Now I don't know why, but I tried changing the bounds box coordinates to another place closer to the airport to test if it's working, but when doing so, the wifi connection doesn't seem to be working anymore.. 

I don't see what's the relation between bounds box and wifi connection.

resmithn, 2024-09-19:
Any messages displayed? What I do is put a few print statements in the code [some are already there, just commented out] to see what is happening. You are right, bounds box shouldn't affect wifi.

MarquisMCMXCIV, 2024-09-19:
Managed to figure out what was the problem with the bounds box.. by removing the # now it constantly says get flights... which is nice, but it still doesn't show any flight information on the display when a flight is overflying my house. I did enlarge the scanning zone to increase the chances of getting something,

resmithn, 2024-09-20:
This is the bound box tool I use.
https://norbertrenner.de/osm/bbox.html
Use mouse wheel to resize the map to your location, then click "draw" upper left.
Bottom right shows the latitude/longitude, add to secrets file paying attention to the exact order of the coordinates.
# area to search for flights: **top latitude, bottom latitude, left longitude, right longitude**
And make sure to include the "-" sign for longitude, if present.
My secrets file looks like this:
'bounds_box' : '27.5892,27.1892,-82.8524,-82.3524'

In the code.py file around statement line 395, uncomment the print for "flight found".
This will let you know that a flight was found.
Also around line 307 should be "print("Flight is called "+flight_number)" which will let you know a flight was found.
If you get the flight number displayed it should scroll the flight info.
Please let me know if you need further help.

MarquisMCMXCIV, 2024-09-21:
Sorry for the delay, just came back from a flight and followed your intructions.

IT WORKS! thank you very much for your help, really apreciated 😁

resmithn, 2024-09-21:
Glad to hear it!

resmithn, 2024-09-21:
Thought I would share this revised code.py file with you. I changed the label positions [centered] and some of the colors. Also added a banner when a flight is found [before the plane animation] and another banner when no flight is found. You can change the "flight found" banner for your location or whatever you like. I just think this looks better than the original layout. Comment if you have any questions and the attached file should run on your setup.
[code.zip](https://github.com/user-attachments/files/17085018/code.zip)

MarquisMCMXCIV, 2024-09-21:
Thanks, I'll try it!

Also if that can be helpful, I THINK when it started working and displaying flights over, the board was not resetting like you guys are talking in the other issue. I don't know if it could have something to do with the internal memory of the board being saturated or something like that.. hope that helps.

resmithn, 2024-09-21:
Yes, very helpful. Thanks.
There is a line in the while loop [towards the bottom of the code] that
will show memory use. You can remove the "#" to have it show up in the
serial monitor.
Usually the code will stop if the board is out of memory.
I was just running the native version and the board was still
disconnecting, I haven't a clue why this all of sudden started happening.

MarquisMCMXCIV, 2024-09-21:
I am still running the original code.. it says memory free 38864.

resmithn, 2024-09-21:
Just checked the code.py modified version that I sent you: memory free
30080.

MarquisMCMXCIV, 2024-09-21:
When it restarts there is a little python flashing in the top left corner. Is there a way to avoid that? Cause apart from that I can't tell it is restarting, it still works fine but this is kind of annoying.

resmithn, 2024-09-21:
Unfortunately no, just the nature of the circuitpython

resmithn, 2024-09-22:
Just an FYI. I had an ESP32-S3 processor [ 8 MB flash, 2 MB of SRAM] board and was playing around with a Word Clock [64x64]. Then yesterday a new issue appeared on this thread, someone trying to run the flight code on an S3, bought by mistake. The S3 handles wifi and requests a little different than the M4. I took the Word Clock code particulars for the S3 and plugged into the M4 Flight code and loaded to the S3. It worked and the surprise is: No disconnects!! So the issue that all are having with the M4 disconnects appears to be something in the code the M4 doesn't like.

MarquisMCMXCIV, 2024-09-22:
I was at my parents house yesterday when I was setting it up first. I had the reset issue during the day, but then when I got home in the evening, I entered my place's wifi information and found out that compared to earlier that day the board was struggling to connect but when it finally did and after finding flights, I found out that it was not resetting anymore. Absolutely no idea why...

resmithn, 2024-09-23:
I have run out of things to try. I have print statements in the code and
best I can tell when it calls "get flights", that is where the disconnect
happens. Course you can no longer see the serial monitor to see what
happened. The try/except block in get flights looks okay, but yet to see a
message from there. So I will just leave it as is. Thanks for your input.

MarquisMCMXCIV, 2024-09-25:
Sorry to bother you again..

Quick update on what I noticed. I cleared the board entirely and reinstalled everything back on it and put it back to the test. 

Yesterday it has been running flawlessly without any resets after flights were detected, and this morning when I woke up it has started to reset again.. so my understanding is that it works fine at the beginning when the board is "empty", but then at some point after some time and after multiple flights are detected, it starts to reset everytime a flight is scanned.

Hope that helps.

resmithn, 2024-09-25:
No worries, thanks for the update! Yes, have experienced the same thing.
Other thing I've noticed is as long as there are flights found one right
after the other and no json error, board doesn't reset.
But I still can't isolate the disconnect in the code on the M4. I have that
modified version running on an S3, no errors and no disconnects, none.
Been trying to help  S3 code? in the issues list [StarTrailMeteor
<https://github.com/StarTrailMeteor>], interesting fwiw.

resmithn, 2024-09-25:
> Tried that code but wifi connection for the s3 code doesn't seem to be
> working on M4. Now the other problem I am facing (I feel I am taking a lot
> of your time lately haha) with the original code, I don't know if it's
> because I live in an apartment and that there is many wifis around, but
> compared to when I first did set it up at my parents house, after the
> resets, it was instantaneously reconnecting to wifi and scanning for next
> flights which was making the reset almost unnoticeable and I could've live
> with it. But now since I've set it at my place, everytime a reset is
> happening, it's not reconnecting and I have to force resets until it does
> using the reset button on the board..
>
> Any idea??

Not a problem. Are you seeing any messages from the wifi function, number
of retries, failed to connect...
It will keep trying to connect up to ten times. How close is the M4 to your
router. You can check your wifi signal with options in the following link.
How to Check/View WiFi Signal Strength in Windows 11? - Winsides.com
<https://winsides.com/how-to-check-view-wifi-signal-strength-in-windows-11/>
But I doubt the signal strength is an issue.

I have the S3 code running and as said before no issues, no connect issues,
it just runs and runs without disconnects.
This is the connect code for the original M4,  you should see messages in
the serial monitor from the print statements.

def checkConnection():
    print("Check and reconnect WiFi")
    attempts=10
    attempt=1
    while (not esp.status == adafruit_esp32spi.WL_CONNECTED) and
attempt<attempts:
        print("Connect attempt "+str(attempt)+" of "+str(attempts))
        print("Reset ESP...")
        w.feed()
        wifi.reset()
        print("Attempt WiFi connect...")
        w.feed()
        try:
            wifi.connect()
        except OSError as e:

print(e.__class__.__name__+"--------------------------------------")
            print(e)
        attempt+=1
    if esp.status == adafruit_esp32spi.WL_CONNECTED:
        print("Successfully connected.")
    else:
        print("Failed to connect.")

MarquisMCMXCIV, 2024-09-25:
I am not home now but basically, just to make sure I understand, your telling me to take that "connect" part and to replace the one in the your S3 code and that it is supposed to be running perfectly?

For my wifi connection issue, looks like it's trying one time only and then the board is crashing and resetting everytime but without any successful connection.

Again thanks for your help!

resmithn, 2024-09-25:
No that code is the M4 code, you said you were running the original code.
Was just pointing out the messages you should see on connect.
Don't change anything.

resmithn, 2024-09-25:
> I don't know if that can be of any help for you, but I asked chat gpt and
> that's what it came out with:
>
> It sounds like the board might be running out of memory or experiencing a
> watchdog timeout, especially since the script is handling large amounts of
> data from the FlightRadar24 API. Let's go through a few potential causes
> and fixes:
> 1. *Memory Management*
>
> Microcontrollers like the MatrixPortal M4 have limited memory, and you're
> working with relatively large JSON files. Even though you are managing
> memory with a fixed json_bytes array, there could still be memory
> fragmentation or exhaustion over time, especially when multiple objects are
> dynamically allocated and freed (such as JSON parsing, networking objects,
> and display labels).
> Solution:
>
>    - *Manually Freeing Memory:*
>    You can attempt to free memory after handling each flight or after
>    every loop iteration by using the gc.collect() function. You are
>    already calling it, but try adding it in more places where memory usage
>    spikes, such as after JSON parsing or network calls.
>
> # Call after handling each flight's datagc.collect()
>
> [... 87 lines cut ...]
>
> By optimizing memory usage, adjusting the watchdog timeout, and
> simplifying the display and network tasks, you can likely resolve the
> crashing issue. Would you like me to go into more detail on how to log and
> debug the crash further?

Thanks, most of those are in the code" w.feed, gc_collect. AI is pretty
smart!
I think we should take this conversion *off* Github. I have some
alternatives to the M4 you may be interested in, or maybe not.
Email me if you are interested, an alternative is running code.py
[modified] on Raspberry Pi4 or Windows with native Python [3.11], not
MicroPython.
my email: ***@***.***
Thanks for your replies.

MarquisMCMXCIV, 2024-09-25:
Your email got automomaticay hidden.. mine is jordanmarquis04 at gmail.com

tschuBBi, 2024-10-31:
Both of you helped me to realise this project for my wife. THANK YOU. =)