smartbutnot/flightportal #27: S3 code?

StarTrailMeteor, 2024-09-20:
Hello!

I thought this project looked cool so I bought the parts and got to work - but it turns out I bought a new S3 portal and not the M4 this this was coded for.

The lovely people on the Adafruit Discord have tried to help with my issues but I'm starting to think they're way above my paygrade, especially as I've never coded before. I believe the main "issue" is that the way that the S3 connects to wifi is different to the M4. So far I've attempted:

resmithn, 2024-09-21:
You are correct: S3 Wifi is different from M4. There are lines at end of code you posted that are not correct, out of place causing syntax error.
What editor are you using? Try MU Editor and use the serial window to review what the code is doing.

resmithn, 2024-09-22:
Made some changes to your code and have it working, with some caveats about password length. Code works with hardcoded ssid and password, but not with variables from the secrets file. That may be an issue for Adafruit support to sort out....or not.
Program has been running over an hour with no disconnects. If there is a json error and same flight you will get blank display for several minutes until valid flight is found. Comment if you are still working on this.

resmithn, 2024-09-22:
Made some changes to your code and is running on S3 with no disconnects.
Please comment if you are still working on this.

StarTrailMeteor, 2024-09-22:
Hello! Yes, sorry, I am still working on this! Apologies for not responding sooner, I didn't realise anyone had commented. I am using Mu but the disconnects were so quick that it took me a while to be able to use the serial to check the issue. I didn't even think about putting it in safemode until someone in the Discord advised me! Can I ask what you changed in order to get it working, please? The current error I am getting is:

Traceback (most recent call last):
  File "code.py", line 454
SyntaxError: invalid syntax

But removing that section of the code leads to the Portal constantly disconnecting, which is why it was added in the first place, so I'm a bit stuck. Any advice would be greatly appreciated!

resmithn, 2024-09-22:
Here is the Code.py file [attached] which has the changes and worked for me. No disconnects and did run that way for several hours. I put comments in the code so that you can see what was added/changed in order for it to work. You can make cosmetic changes to the code as far as spacing and what is displayed on the matrix. Any questions, just comment and will try to help.
[code.zip](https://github.com/user-attachments/files/17091287/code.zip)

StarTrailMeteor, 2024-09-22:
That is so unbelievably kind of you, thank you! It's amazing how helpful this community has been so far - especially as I'm sure the questions I am asking are very basic!

I've had a quick run of the code, and unfortunately I think I might have messed something up. When I ran it initially I got the error (from Mu) stating:

File "code.py", line 8, in <module>
  File "adafruit_matrixportal/matrixportal.py", line 33, in <module>
  File "adafruit_matrixportal/network.py", line 31, in <module>
ImportError: no module named 'neopixel'

So then I thought I'd missed the neopixel library, so I added then and subsequently got:

Traceback (most recent call last):
  File "code.py", line 8, in <module>
  File "adafruit_matrixportal/matrixportal.py", line 33, in <module>
  File "adafruit_matrixportal/network.py", line 31, in <module>
  File "adafruit_seesaw/neopixel.py", line 25, in <module>
ImportError: seesaw neopixel being imported from unexpected location - is seesaw neopixel use intended?

Am I missing something potentially very obvious? I am also still getting some disconnecting but that could potentially have been the cable I'm using (had to borrow one to run the code quickly!).

resmithn, 2024-09-23:
Looks like your library folder is out of sync or not complete.
Attached is the library list I am using. Also the S3 requires CircuitPython
8.2. also what I have loaded.
Here is link to Circuit Python 8.x libraries.
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/download/20240922/adafruit-circuitpython-bundle-8.x-mpy-20240922.zip
This is my library list:
[image: Screenshot 2024-09-19 121322.jpg]

StarTrailMeteor, 2024-09-23:
I can't check the code right now but I just wanted to say thank you again - you've been really helpful and it's very much appreciated. I can't see your screenshot for some reason but I assume the list of libraries isn't much different from the ones used in the original code?

Also thank you for the comments! As frustrating as learning this has been it's also been incredibly fun so I am pretty excited to see what you've changed and why. Hopefully at some point I'll be able to understand it fully!

resmithn, 2024-09-24:
> Okay so I couldn't get your screenshot of the libraries to load, so I had
> a look around and used the screenshot you shared on another thread.
> After adding them all I get the error:
> File "adafruit_matrixportal/matrixportal.py", line 32, in
> ImportError: no module named 'adafruit_portalbase'
> And then when adding the adafruit_portalbase library I get the error:
> I am completely baffled. I'm using 8.2 - I was initially using the most
> recent release (9.1?) but I changed it on your recommendation. Would love
> to hear if you (or anyone else!) have any thoughts!

Please delete the folders and files in your "lib" folder [ on the board
drive] and reload from the  link I sent you, for circuitpython bundle 8.x.
These errors indicate the lib [on the board drive] files/folder are not in
sync.

StarTrailMeteor, 2024-09-24:
I did - those are the libraries I am currently using that are causing the errors, unfortunately. Are you using the same library list in the screenshot from the need help thread? Maybe I've added something I shouldn't?

resmithn, 2024-09-25:
When the M4 code was modified to run on an S3 I was using a 32x64 display
and it worked fine. I still don't understand why you are seeing errors,
there has to be something amiss with your library(s).
Then I realized you are using a 64x64 display. Does not matter, code should
still run, but the display won't be correct. The 64x64 uses 5-address
multiplexing system - 32x64 uses 4-addresses. This means the code for 64x64
requires logic for the 5 addresses, but is not needed for 32x64.
Take a look at the Word Clock code in the link below. It runs on S3 with a
64x64 display. You may want to try it and see if it works on your setup.
RGB Matrix Word Clocks | Adafruit Playground
<https://adafruit-playground.com/u/VPTechOps/pages/rgb-matrix-word-clocks>

In order to run the S3 with a 64x64 display the modified flight code would
need the additional address logic for the display to look correct.
I ran the Flight Code on a 64x64 display this morning and you can see the
gap for the second line.

[image: FlightRadar_64x64.jpg]
This means the Matrix display object [in the code.py file] is missing the
address info. One can probably modify the S3 code with address info, but at
this point I'm not sure how to do that. Will not be easy.
I also attached a zip.file with the code.py [named codeS3.py] and the
[lib]rary folder. This is the code I ran this morning that is showing the
flight data in the picture above.
I would recommend that you obtain a 32x64 display and use the native code
from the Github project [or the S3 code.py].

resmithn, 2024-09-25:
zip file
[lib.zip](https://github.com/user-attachments/files/17131862/lib.zip)
[image]

resmithn, 2024-09-25:
Zip File and 64x64 Image
[image]
[lib.zip](https://github.com/user-attachments/files/17131935/lib.zip)

StarTrailMeteor, 2024-09-25:
So, I didn't think it was an issue with the size of the display because I am currently using a 64x32, so I went back and re-did all my libraries according to the file you provided and this time it worked! For literally the first time since I started I saw something on the display, which is amazing. The only issue is - the text/images don't seem to be right. Everything is displaying as it should be (as far as I can tell) but the text pixels seems to have big gaps and it seems overlaid on top of each other. I thought it was an issue with the font or size, but I think some of the banners are actually running over each other as well. Specifically the "Flight Radar for" one, I think. I'm not sure why this is happening because the clear display is working. Maybe it's the timing?

resmithn, 2024-09-25:
Let's take this issue off the Github thread. Will try to help you get
things resolved.
My email is resmithn at gmail.com. We can talk tomorrow.
Thanks.

resmithn, 2024-09-26:
Still not clear on which display you are using. Your original post said 64x64, last post you said 64x32. All indication is that you are using 64x64 based on the gaps and overlaid pixels you are seeing. I explained in post before that 64x64 will not work with the code you have because of the 64x64 addressing, which is not there. The code you have will work with a 32x64 display.
Please send me an email [address in last post] if you want to comment further.

keanenwold, 2024-12-30:
`This is my library list: [image: Screenshot 2024-09-19 121322.jpg]`

@resmithn, Could you please share this image in GitHub so we can see it? thanks!

resmithn, 2024-12-30:
Library Image
[image]

gjchester, 2025-10-02:
Sorry jumping into another thread  :) 

My M4 on V8 works but resets every so often , so I want to try and fix it..  

I tried out my S3 too, 

Boot_Out says...

```
Adafruit CircuitPython 8.2.0 on 2023-07-05; Adafruit MatrixPortal S3 with ESP32S3
Board ID:adafruit_matrixportal_s3
UID:8273F2CE2FC5
```
This give  me the error

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 49, in <module>
  File "adafruit_requests.py", line 366, in __init__
  File "adafruit_connection_manager.py", line 395, in get_connection_manager
TypeError: unsupported type for __hash__: 'SocketPool'

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

All the libraries downloaded from your link 

[image]

I'm also grateful for any suggestions, I an use the M3 but I;'d rather it not reset all the time as it flashes code on the display for a second, which is detracting and a bit disappointing if there no plane to show for it  :)