smartbutnot/flightportal #31: Brutal Newbie Quest - Files lost after upload / Not loading?

fnhabach, 2024-11-25:
I have a lot of great things about this project and couldnt wait to re-create this asap.
Unfortunately, I am an absolute entire newbie when it comes to this sort of stuff.

I have managed to purchase all the equipment, have connected the matrix portal, double clicking the reset button and seeing the drive pop up on my windows. Once I copied over the 2 files (secrets and code), it shows as being in there, but once I click the reset button again - its all gone!

Am I supposed to be installing some sort of bootloader? Is there a different method of uploading these files? 
Any advice would be greatly appeciated! 

Below a picture of what I see when I double click reset incl. what the device shows. 

[image]
[image]

iminspaaace, 2024-12-07:
You are missing a step to install CircuitPy - this is what runs the code. Make sure you are following the guide here: https://learn.adafruit.com/adafruit-matrixportal-m4/prep-the-matrixportal You have to click "next" at the bottom right on each page.

iminspaaace, 2024-12-07:
side note, I haven't been successful on anything later than CircuitPy 7.3.3 but I'm running a version of the code from early 2023, not the latest. Not sure what's going on.

fnhabach, 2024-12-07:
@iminspaaace thank you soo much! I installed the latest version just as you had suggested, I saw CircuitPy and I copied over the files. Now, I dont see the Adafruit colours upon restarting or clicking the reset button. Now, I just see occassional blinking lights on the chip and thats it. 

Really sorry about this - but your guidance is seriously appreciated. 
[image]

iminspaaace, 2024-12-11:
I have not had luck with the latest version of CircuitPy and this code, it seems to crash. I am using 7.3.3 with marginal stability. Sorry I can't help more.