smartbutnot/flightportal #37: Please Help a newcomer

Eyecodagain, 2025-01-08:
I have tried to install using 8.1 and 9.1 and i have also used to associated libraries but no joy

Matrix M4

Currently trying 9.1 and Lib Bundle 9.x

It constantly restarts and I get the below

Traceback (most recent call last):
  File "code.py", line 33, in <module>
  File "secrets.py", line 12
SyntaxError: invalid syntax

I also have these in my library 
[image]

Doe anybody have any suggestions pls?

benfugate, 2025-02-11:
This repo runs on CircuitPython 8.X.X, so you will want version 8.2.10. Or, if you're set on 9.X, there is an open PR and you can use the latest commit from that: https://github.com/smartbutnot/flightportal/pull/35

benfugate, 2025-02-11:
@Dom-Evans 
If you have CircuitPy 8.2.10 installed, you will need the corresponding .mpy libraries in your "lib" folder.

You can grab the latest bundles here: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases
At the time of writing, you'll want the latest 8.x libraries: adafruit-circuitpython-bundle-8.x-mpy-20250211.zip

Open this ZIP download and navigate to the "lib" folder. You will want to grab the files defined in this projects README: https://github.com/smartbutnot/flightportal/blob/main/README.md?plain=1#L25-L32

You will also need adafruit_ticks.mpy

Dom-Evans, 2025-02-11:
Thanks @benfugate - Looking at that ZIP now, I have added in the files and any others which the script is returning when there is an issue. I am now faced with this? Apologies, I am not the most literate when it comes to this! 

[image]

benfugate, 2025-02-11:
@Dom-Evans 
Happy to assist - Line 53 has two things that _should_ be strings, FLIGHT_SEARCH_HEAD, BOUNDS_BOX, and FLIGHT_SEARCH_TAIL.
Any chance you left BOUNDS_BOX empty in the [secrets.py](https://github.com/smartbutnot/flightportal/blob/main/secrets.py) file? Or, on the same wavelength, ensure you dropped that secrets.py file alongside code.py.

Dom-Evans, 2025-02-11:
That information (bounds box) being on the secrets file and not the code file?

benfugate, 2025-02-11:
Correct, code.py should not require any modifications, which appears to be the issue. FLIGHT_SEARCH_TAIL appears to have been accidentally deleted, I just noticed from your screenshot.
https://github.com/smartbutnot/flightportal/blob/main/code.py#L57

benfugate, 2025-02-11:
Nope, I'm just dumb and cant read. Hmm. It's definitely BOUNDS_BOX since those two are slowing there.

benfugate, 2025-02-11:
I would still reset your code.py to match this repos master version. Your line numbers don't line up, which lead me to think there still may have been an unintentional modification.

Dom-Evans, 2025-02-11:
No problem, just reset the code as you described. Think it was because I was trying to search multiple things online.

IT HAS WORKED! Thank you so much, now just to change the bounds to my local area! Thanks again for your help! Do you have a buy me a coffee link or anything for me to say thanks! 

Now just need an aircraft to fly over...might have to expand the box 😆

benfugate, 2025-02-11:
Great to hear it's working! No coffee needed, I appreciate it though! :tada:

Dom-Evans, 2025-02-11:
Sorry noticed that whilst it works for a few flights, every 10 minutes after a flight has been shown the matrix portal light will flash green, cause the RGB panel to flash a few pixels (nothing readable/only a few lights) and then display a red light on the portal. It the repeats by doing a few green flashes and repeats. This is whilst other aircraft are flying within the bounds box.

I have found to get it working again, I usually press the reset button on the portal or turn off the power and back on again

Anyone else had this issue or know the cause?