smartbutnot/flightportal #39: Is there any chance the API changed for FlightRadar?

nnoble3945, 2025-03-04:
I'm able to get the code running, and am trying to troubleshoot the results of get_flights(). The JSON that is returned for `response` is:

`{'full_count': 15742, 'version': 4}`

The if statement within get_flights() therefore never returns TRUE as only 2 elements are returned. 

I'm a beginner, so apologies if I missed a step in the setup.

resmithn, 2025-03-05:
Flight Radar server/website was down in the afternoon on 03/04/2025. Website came back in the evening but appears to still be issues. This is error I am seeing with the script on the matrix and not sure how to correct it.

Returning Flight ID
395ce850
New flight 395ce850 found, clear display
Flight Number in Get Flight Details
https://data-live.flightradar24.com/clickhandler/?flight=
395ce850
This is in Get Flight Details and e is the error
Expecting value: line 1 column 1 (char 0)
error loading details, skip displaying this flight.

Getting similar message from a Windows version of the script.

{'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:106.0) Gecko/20100101 Firefox/106.0', 'cache-control': 'no-store, no-cache, must-revalidate, post-check=0, pre-check=0', 'accept': 'application/json'}
This is in Get Flight Details and e is the error
Expecting value: line 1 column 1 (char 0)
error loading details, skip displaying this flight

Will try research further but looks like issue with the Flight Radar website: https://data-live.flightradar24.com/clickhandler/?flight=
395ce850 OR the data that is being returned.

resmithn, 2025-03-05:
Yes,  website went down 3/04, came back but something has changed in the json data returned for the Get Flight Details. Just started to look at the code. I Googled the error [Expecting value: line 1 column 1 (char 0)], still looking at that, but don't see much that will help. Will post if any answer found. This will be an issue for the Author to look into.

resmithn, 2025-03-06:
Looking into the error further found the HTTP Status code is 403, i.e., not authorized.
Which then produced this error:

This is in Get Flight Details and e is the error
Expecting value: line 1 column 1 (char 0)
error loading details, skip displaying this flight

This is code thread up to Get Flight Details:
https://data-cloud.flightradar24.com/zones/fcgi/feed.js?bounds=27.6892,27.0692,-82.9624,-82.2524&faa=1&satellite=1&mlat=1&flarm=1&adsb=1&gnd=0&air=1&vehicles=0&estimated=0&maxage=14400&gliders=0&stats=0&ems=1&limit=1
In Start up
Get flights...
full_count 13429
version 4
395f1fe7 ['', 27.4362, -82.7193, 281, 43000, 400, '', 'F-BDWY1', 'GLF5', '', 1741259817, '', '', '', 0, -64, 'GLF5', 0, '']
Returning Flight ID
395f1fe7
New flight 395f1fe7 found, clear display
Flight Number in Get Flight Details
**https://data-live.flightradar24.com/clickhandler/?flight=395f1fe7**
Headers:
{'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:106.0) Gecko/20100101 Firefox/106.0', 'cache-control': 'no-store, no-cache, must-revalidate, post-check=0, pre-check=0', 'accept': 'application/json'}

Status Code
403

I tried the url for the flight in a Chrome Browser [highlighted in BOLD above], page loads but on Inspect the Console tab says:
Failed to load resource: the server responded with a status of 404 ()

No idea on how to correct this, but it appears FlightRadar has shut this link down.
Hopefully someone can confirm this.

felker, 2025-03-07:
I had my setup working as recently as May 2024, but took it offline thereabouts. I started revisiting this yesterday and updating my fork of the code. Was https://data-cloud.flightradar24.com/zones/fcgi/feed.js working as recently as last month? Or has it been broken longer?

Perhaps these developments are related to this Oct 2024 announcement of a paid API? https://www.flightradar24.com/blog/b2b/flightradar24-api/

resmithn, 2025-03-07:
Link stopped working on 3/04/25 late evening. Website was down for several hours late afternoon, when it came back, the link stopped working.

nnoble3945, 2025-03-07:
I'll do a little bit of research. I noticed pulling https://data-cloud.flightradar24.com/zones/fcgi/feed.js? works, but it does not appear to have current data (or at least it's incomplete). Once you add the bounds and FLIGHT_SEARCH_TAIL it returns just the full_count and version parts of the JSON. I think something definitely changed. 

I updated the entire code to use a different API from the OpenSkyNetwork. It pulls less information (i.e. does not have arrival / departure airports, had to create a manual dictionary of callsign to airline names, etc.) but it does the trick for to at least learn the code a bit.

smartbutnot, 2025-03-08:
Hi all, author here - sorry, I don't have much time to look at this project these days, I had a kid since I wrote it! Thanks to everyone who's kept it working in the meantime! Anyway, had a very quick look, and while I was doing that, it seems like it's responding on that URL again? From what I can tell, right now, it's working.

resmithn, 2025-03-08:
Thank You, working okay here.