smartbutnot/flightportal #41: Newby in way over my head?!?

bskinnell, 2025-05-23:
I came across this flightportal project from 'smartbutnot' and was super excited to give it a shot.  I ordered everything needed from Adafruit.  After receiving delivery, I followed the instructions; downloaded the most current UF2 version (9.2.7) and got the board (Matrix Portal M4)  setup without issue, copied the code.py file and the necessary libraries to the CIRCUITPY drive.  Going through the instructions, I understand that there seems to be a significant change related to the 'secrets' file; it appears that the newer UF2 version uses the settings.toml instead (which I have done).  Here is where I have fallen flat on my face.....I have spent several hours 'googling' and tried everything that I can think of changing the code.py file and cannot get anything to work.  

I would REALLY appreciate anyone's help to get up and running (or maybe just walking).  I would really hate to have just wasted money on all of this for nothing.

Thank you in advance for any consideration with this request.

gjchester, 2025-10-02:
Possible stealing someone else threat but mines in the same state.

Got it all working on Circuit Python 8.x, and backed it up just in case then tried to move to Circuit Python 9.  I think the error is that the code is still looking for secret.py  (line 32 maybe) 

[image]

Issue 33 / 34 / 35 refers to updating the code, but they don't look to be merged,  if my limited understanding of GitHub messages is right,  so anyone who isn't familiar with making the changes may have a V8 code trying to run on V9.

resmithn, 2025-10-02:
CircuitPython has shifted away from using secrets.py for storing network credentials and other sensitive information. The recommended practice now involves using a settings.toml file.  You may want to try using settings.toml instead of secrets file if you are running V9. Also, just a comment, may be better to stay on V8 since that is what the original code was written for. I have V8.2 and runs okay.

gjchester, 2025-10-02:
I get that and I mine's current working on 8.2.10.

I think the problem is ore if you try and follow this your initial work to prep the board will lead you to try and use V9 or V10 UF2, but the codes still written expecting V8.

I plan to try using the code by compileByTheMile on his fork (https://github.com/compileByTheMile/flightportal) tonight to try and get mine to CircuitPython 9 , knowing that I have all the files to revert it back to V8 in a folder if I need them