smartbutnot/flightportal #42: Connects once, cant find SSID thereafter

easo345, 2025-08-07:
The device finds a flight right after you power it on, then seems to be unable to find the SSID thereafter. I've attached a screenshot of the serial monitor log:

[image]

Running 8.2.10 uf2 and bundle 8.x-mpy-20250211

Has anyone else had this problem?

rangus269, 2025-08-13:
Experiencing a similar issue. Almost appears to time out and then just get flashing red on the MatrixPortal

rangus269, 2025-09-07:
I used #34 from @CompileByTheMile and have had no issues since. I'm considering a branch to make this clearer, as it took me quite some time to locate and find that code.

https://github.com/smartbutnot/flightportal/tree/521e7dfa3020a56c30eefeccaa872cb5d4e672f2